shadow-robot/sr-ronex #180: Reading Analog Input

blackcoder, 2015-05-31:
I am trying to use the RONEX modules for the first time. I plugged the cables, powered the adapter, and launched the driver but I get an error : 
``` 
root@kuri-CELSIUS-M720:~/catkin_ws_generic/src/sr-ronex/sr_ronex# roslaunch sr_ronex_launch sr_ronex.launch
... logging to /home/kuri/.ros/log/c045a8b2-0792-11e5-a95e-e03f4992b197/roslaunch-kuri-CELSIUS-M720-4605.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://10.0.0.197:56239/

SUMMARY
========

PARAMETERS
 * /robot_description: <?xml version="1....
 * /rosdistro: indigo
 * /rosversion: 1.11.13

NODES
  /
    ronex (ros_ethercat_loop/ros_ethercat_loop)
    ronex_ctrl_spawner (sr_ronex_controllers/load_passthrough_controllers.py)

auto-starting new master
process[master]: started with pid [4620]
ROS_MASTER_URI=http://10.0.0.197:11311

setting /run_id to c045a8b2-0792-11e5-a95e-e03f4992b197
process[rosout-1]: started with pid [4633]
started core service [/rosout]
process[ronex-2]: started with pid [4650]
GNU gdb (Ubuntu 7.7.1-0ubuntu5~14.04.2) 7.7.1
Copyright (C) 2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /opt/ros/indigo/lib/ros_ethercat_loop/ros_ethercat_loop...(no debugging symbols found)...done.
Starting program: /opt/ros/indigo/lib/ros_ethercat_loop/ros_ethercat_loop -i eth0 -r /robot_description __name:=ronex __log:=/home/kuri/.ros/log/c045a8b2-0792-11e5-a95e-e03f4992b197/ronex-2.log
[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
process[ronex_ctrl_spawner-3]: started with pid [4656]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
[New Thread 0x7fffeefcf700 (LWP 4660)]
[New Thread 0x7fffee7ce700 (LWP 4661)]
[New Thread 0x7fffedfcd700 (LWP 4662)]
[New Thread 0x7fffed7cc700 (LWP 4667)]
[New Thread 0x7fffecfcb700 (LWP 4668)]
[New Thread 0x7fffdffff700 (LWP 4670)]
[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
[New Thread 0x7fffdf5f2700 (LWP 4674)]
[New Thread 0x7fffdedf1700 (LWP 4676)]
[ WARN] [1433076260.228414267]: Using device 'RoNeX - Bridge module mk3' with product code 33554432
[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
[ WARN] [1433076260.466759102]: Using device 'RoNeX - General I/O module mk3' with product code 33554433
[ INFO] [1433076260.468728556]: Using EC_QUEUED
[ INFO] [1433076260.468809362]: First FMMU (command) : Logical address: 0x00010000 ; size:  44 bytes ; ET1200 address: 0x00001000
[ INFO] [1433076260.468901134]: Second FMMU (status) : Logical address: 0x0001002C ; size:  30 bytes ; ET1200 address: 0x000010B0
[ INFO] [1433076260.468974607]: Finished constructing the SrBoardMk2GIO driver
[ INFO] [1433076260.558359018]: Device #00: Product code: 33554432 (0X02000000) , Serial #: 1409326955 (0X54009F6B)
[ INFO] [1433076260.558461894]: Device #01: Product code: 33554433 (0X02000001) , Serial #: 1409579150 (0X5404788E)
[New Thread 0x7fffde008700 (LWP 4688)]
[ INFO] [1433076260.575090755]: Adding a general_io RoNeX module to the hardware interface: /ronex/general_io/1409579150
[New Thread 0x7fffdd807700 (LWP 4691)]
[ INFO] [1433076260.583796052]: Added Ethernet port eth0
[New Thread 0x7fffdd006700 (LWP 4699)]
[INFO] [WallTime: 1433076262.346868] Waiting for the ronex to be loaded properly.
[INFO] [WallTime: 1433076262.356397] Waiting for controller_manager services: list/load/switch
[INFO] [WallTime: 1433076262.363767] Starting controllers: ['ronex_1409579150_passthrough']
[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
[ronex_ctrl_spawner-3] process has finished cleanly
log file: /home/kuri/.ros/log/c045a8b2-0792-11e5-a95e-e03f4992b197/ronex_ctrl_spawner-3*.log

```

and when I display the topics published by the driver I only get this:

```
kuri@kuri-CELSIUS-M720:~/catkin_ws_generic$ rostopic list
/diagnostics
/ronex/general_io/1409579150/command/digital/0
/ronex/general_io/1409579150/command/digital/1
/ronex/general_io/1409579150/command/digital/10
/ronex/general_io/1409579150/command/digital/11
/ronex/general_io/1409579150/command/digital/2
/ronex/general_io/1409579150/command/digital/3
/ronex/general_io/1409579150/command/digital/4
/ronex/general_io/1409579150/command/digital/5
/ronex/general_io/1409579150/command/digital/6
/ronex/general_io/1409579150/command/digital/7
/ronex/general_io/1409579150/command/digital/8
/ronex/general_io/1409579150/command/digital/9
/ronex/general_io/1409579150/command/pwm/0
/ronex/general_io/1409579150/command/pwm/1
/ronex/general_io/1409579150/command/pwm/2
/ronex/general_io/1409579150/command/pwm/3
/ronex/general_io/1409579150/command/pwm/4
/ronex/general_io/1409579150/command/pwm/5
/ronex/general_io/1409579150/parameter_descriptions
/ronex/general_io/1409579150/parameter_updates
/ronex/general_io/1409579150/state
/ros_ethercat/motors_halted
/rosout
/rosout_agg
```

Anything that I am missing here ? This module is supposed to be very simple to use ...

ugocupcic, 2015-06-01:
As far as I can see there's no error in that log you pasted. To get the inputs from your RoNeX you need to subscribe to the state topic. In your case: `rostopic echo /ronex/general_io/1409579150/state` should do it. Let me know if it works!

Cheers,

blackcoder, 2015-06-01:
Thanks @ugocupcic I figured out that, I can successfully read the analog data. I can only see, however, 6 analogue channels, not 12, how can I access the other 6 ?

Cheers

ugocupcic, 2015-06-02:
Great to hear that. I'll close this issue then. 

Regarding 6 vs 12 channels, it depends how many stackers (the boards with the cable connections) you have plugged in your RoNeX. It will pick them up automatically.

closed 2015-06-02 by ugocupcic

Rocketmagnet, 2015-06-02:
But make sure you power it down before plugging in the next stacker.