shadow-robot/sr-ronex #183: Failed to find ronex devices in parameter server

rbrth, 2015-06-03:
Hi all,

I connected my Ronex device and I am trying to launch the Ronex driver with:

$ roslaunch sr_ronex_launch sr_ronex.launch ethercat_port:=eth1

Although it finds the device, it fails to find the Ronex module in the parameter server:

root@cortana:~# roslaunch sr_ronex_launch sr_ronex.launch ethercat_port:=eth1
... logging to /home/ruud/.ros/log/8b86041a-09c6-11e5-b125-c8bcc8ebb9e1/roslaunch-cortana-13685.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://cortana:50847/

SUMMARY
========

PARAMETERS
 * /robot_description: <?xml version="1....
 * /rosdistro: indigo
 * /rosversion: 1.11.10

NODES
  /
    ronex (ros_ethercat_loop/ros_ethercat_loop)
    ronex_ctrl_spawner (sr_ronex_controllers/load_passthrough_controllers.py)

auto-starting new master
process[master]: started with pid [13700]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 8b86041a-09c6-11e5-b125-c8bcc8ebb9e1
process[rosout-1]: started with pid [13713]
started core service [/rosout]
process[ronex-2]: started with pid [13730]
process[ronex_ctrl_spawner-3]: started with pid [13747]
[ WARN] [1433318406.778896806]: Using device 'RoNeX - Bridge module mk3' with product code 33554432
[ INFO] [1433318406.896560732]: Device #00: Product code: 33554432 (0X02000000) , Serial #: 1409327303 (0X5400A0C7)
[ INFO] [1433318406.904184404]: Added Ethernet port eth1
[INFO] [WallTime: 1433318409.459327] Waiting for the ronex to be loaded properly.
[INFO] [WallTime: 1433318414.579294] Failed to find ronex devices in parameter server
[ronex_ctrl_spawner-3] process has finished cleanly
log file: /home/ruud/.ros/log/8b86041a-09c6-11e5-b125-c8bcc8ebb9e1/ronex_ctrl_spawner-3*.log

ugocupcic, 2015-06-03:
@rbrth it seems like you only have the bridge connected, not the general I/O board.

Basically you seem to only have the 2 first boards (counting from the board with the big ethernet connector) on the pic below. You need to have the 3rd board attached as well as shown in this pic.

[image]

Let me know if this answers your question.

rbrth, 2015-06-03:
@ugocupcic , it was attached, so that is not the cause unfortunately.

RobShadow, 2015-06-03:
Can you clarify how many PCBs you have attached - 3 or 4?

rbrth, 2015-06-03:
Dear Rob,

I currently have 3 boards attached, as I do not require the SPI one yet.

Thank you for your quick reply.

Best,

Ruud

RobShadow, 2015-06-03:
Hi Ruud,
Did you receive 3 boards only, or 4 with your order?

rbrth, 2015-06-03:
Hi Rob,

I received 4 boards with my starter kit.

I now have attached the 4th board and I see the general I/O in the ros parameter server and the error diasspearing in the console.

Apperently all boards have to be connected? Perhaps a good idea to add this information to the wiki, as I thought the modular system would be also functional without the module I did not need at this moment.

Best,

Ruud

RobShadow, 2015-06-03:
The 3rd and 4th board combine to make the GIO module. The 4th board (GIO Stacker) feeds back into the 3rd board (GIO Node) to tell it what is plugged in.

You're right, this absolutely does need to be mentioned, and I'm very surprised it isn't. We'll have a read through and make it much clearer.

Thanks, and good luck!

closed 2015-06-03 by toliver

rbrth, 2015-06-03:
Fixed!

I now see that both boards are GIO. I falsly presumed one board was SPI in the starter kit and only used the one that is displayed in the 'General I/O Module Specification' document.