shadow-robot/sr-ronex #184: Only 6 of the 12 IOs, cannot toggle over command line

rbrth, 2015-06-03:
Hey,

I am trying to toggle a digital IO over the command line, but fail to do so:

ruud@cortana:~$ rostopic pub /ronex/general_io/1409581363/command/digital/0 std_msgs/Bool '{data: true}'

I also tried values like True/true/1/False/false/0 to make sure that was not the issue.

ruud@cortana:~$ rostopic echo /ronex/general_io/1409581363/state
header: 
  seq: 143385
  stamp: 
    secs: 1433339986
    nsecs: 719700198
  frame_id: ''
digital: [False, False, False, False, False, False]
analogue: [0, 0, 0, 0, 0, 2]
pwm_clock_divider: 20
input_mode: [True, True, True, True, True, True]
---

Furthermore, only 6 out of the 12 digital IOs show up, wheras there should be 12 (/ronex/general_io/12/command/digital/0 (0-11) [std_msgs/Bool]) :

ruud@cortana:~$ rostopic list
/diagnostics
/ronex/general_io/1409581363/command/digital/0
/ronex/general_io/1409581363/command/digital/1
/ronex/general_io/1409581363/command/digital/2
/ronex/general_io/1409581363/command/digital/3
/ronex/general_io/1409581363/command/digital/4
/ronex/general_io/1409581363/command/digital/5
/ronex/general_io/1409581363/command/pwm/0
/ronex/general_io/1409581363/command/pwm/1
/ronex/general_io/1409581363/command/pwm/2
/ronex/general_io/1409581363/parameter_descriptions
/ronex/general_io/1409581363/parameter_updates
/ronex/general_io/1409581363/state
/ros_ethercat/motors_halted
/rosout
/rosout_agg

Thanks!
Ruud

toliver, 2015-06-03:
You probably need to define that pin as output first. Please thy to do that through the dynamic_reconfiguration plugin. By running:

```bash
rqt
```
And then opening the plugin.

RobShadow, 2015-06-04:
Hi Ruud, 
Did you manage to sort this by using the rqt command for the configuration gui? 
--
Sent from myMail app for Android

rbrth, 2015-06-04:
@toliver : setting the pin again as output works! THe information on ( https://github.com/shadow-robot/sr-ronex/wiki/GIO-Module-PWM-LED-%28CL%29) is outdated though, because you have to fill in all the fields of the message for it to work. Otherwise the command line throws an error.

root@cortana:~rosservice call /ronex/general_io/1409581363/set_parameters "config:
  bools:
  - {name: 'input_mode_0', value: false}
  ints:
  - {name: '', value: 0}
  strs:
  - {name: '', value: ''}
  doubles:
  - {name: '', value: 0.0}
  groups:
  - {name: '', state: false, id: 0, parent: 0}"

ugocupcic, 2015-06-04:
Great thanks for pointing that out, I reported issue #186 and will try to fix this doc as soon as possible.

rbrth, 2015-06-04:
Still only 6 out of the 12 digital IOs show up, wheras there should be 12 (/ronex/general_io/12/command/digital/0 (0-11) [std_msgs/Bool]) :

ruud@cortana:~$ rostopic list
/diagnostics
/ronex/general_io/1409581363/command/digital/0
/ronex/general_io/1409581363/command/digital/1
/ronex/general_io/1409581363/command/digital/2
/ronex/general_io/1409581363/command/digital/3
/ronex/general_io/1409581363/command/digital/4
/ronex/general_io/1409581363/command/digital/5
/ronex/general_io/1409581363/command/pwm/0
/ronex/general_io/1409581363/command/pwm/1
/ronex/general_io/1409581363/command/pwm/2
/ronex/general_io/1409581363/parameter_descriptions
/ronex/general_io/1409581363/parameter_updates
/ronex/general_io/1409581363/state
/ros_ethercat/motors_halted
/rosout
/rosout_agg

ruud@cortana:~$ rostopic echo /ronex/general_io/1409581363/state
header:
seq: 143385
stamp:
secs: 1433339986
nsecs: 719700198
frame_id: ''
digital: [False, False, False, False, False, False]
analogue: [0, 0, 0, 0, 0, 2]
pwm_clock_divider: 20
input_mode: [True, True, True, True, True, True]

toliver, 2015-06-04:
This might be due to the order in which you have plugged the 2 boards of the GPIO module.

The node (the one with the PSoC chip) should go before (i.e. closer to the power supply board) the extension board.

RobShadow, 2015-06-04:
This board should be in 3rd place (moving away from ethernet connector)
[image]

This should be in 4th place, ie on the top of the stack
[image]

rbrth, 2015-06-04:
@RobShadow Yes that worked! I falsely assumed, due to the modularity of the system, the order wouldn't make a difference. Also, I did not read it anywhere (yet).

Thanks,
Ruud

closed 2015-06-04 by rbrth